OttoDIY/blockly #157: Linux packages for latest releases (post 1.4.2)?

casusbelli, 2022-03-07:
Hi!
Is there a location providing newer packages of blockly than 1.4.2 for Linux?

We've been running the OS X version of blockly, where we can upload and run stuff on our otto but there are compile issues with 50-60% of the blockly blocks. 
We now installed a linux system on an old laptop which builds the code fine but so far i'm unable to upload code to our otto.
While checking this repo i found that the installed blockly release is somewhat outdated (1.4.2) which is why i'm looking for packages of the later releases.
Thanks!

cparrapa, 2022-03-09:
Hi @casusbelli 

We put them here https://github.com/OttoDIY/blocklyLinux then you need to remplace all the www folder from this repo to keep it updated.

Ideally we want to merge to this repo so it is all together but we don't know how to exactly create the installers without interfering with other OS, so help need it here.

OS X version which one are you using and what blocks specifically have bugs? Is this something that works on Win?

rejas, 2022-03-11:
Hi @cparrapa is there a task list of stuff to do to get linux support merged in here? Or how can I help?

cparrapa, 2022-03-11:
Hi @rejas 

To be honest I don't know were to start, maybe easier to create the proper scripts for packing on Linux  in the packgage.json file?
https://github.com/OttoDIY/blockly/blob/master/package.json

Long story short, a guy made the Linux version that is one of the branches in this repo then he updated and we put here https://github.com/OttoDIY/blocklyLinux since but He made many changes to be able to work as the Windows version and does not use Arduino CLI, so that is why we kept separate until we figure out how to merge into one Software, that is technically possible since we are using Electronjs.

I am not familiar with Linux for developing, so any support is greatly appreciated. Thanks for your help.

logix5, 2022-03-11:
Hi  @rejas

I would like to have a linux version too using the same branch than windows
and linux. I think the solution is near, maybe using the current version of
index.js is ok, because I think Linux calls the arduino-cli as MAC, then
maybe it works directly :-) . We only will need to create the final Linux
version and write down the steps to  install arduino-cli for Linux.

Maybe that is all and we only need to test to check this way is ok . If you
want to test it would be interesting , because i have only a
windows computer

Important in github we need to write-down in the main the steps to create
the Windows, MAc and Linux version

Thanks in advance

-- 
Óscar Ferruz Romero
*Socio fundador.*
*Tel. 620.402.168*

 C/ Artajona, 17.  28039 - Madrid.
www.logix5.com

www.twitter.com/Logix5_SS <http://www.twitter.com/logix5_SS>

www.facebook.com/logix5

www.instagram.com/logix5_smart_solutions

Conforme dispone el Reglamento (UE) 2016/679 del Parlamento Europeo y del
Consejo, de 27 de abril de 2016, relativo a la protección de las personas
físicas en lo referente al tratamiento de datos personales y a la libre
circulación de estos, le informamos que los datos personales facilitados y
recogidos en este e-mail y sus documentos adjuntos serán tratados e
incluidos automatizadamente en los ficheros de datos personales bajo
responsabilidad de LOGIX5 SMART SOLUTIONS, S.L. B-86129699, donde se
recogen y almacenan los datos personales con la finalidad exclusiva de
formar parte en los procesos de selección de personal, bolsa de trabajo,
contratación o gestiones relacionadas con temas laborales que se lleven a
cabo. El interesado otorga su consentimiento para el tratamiento de sus
datos personales con la finalidad anteriormente mencionada. En el supuesto
de producirse alguna modificación de sus datos personales, le solicitamos,
nos lo comunique por escrito con la única finalidad de mantenerlos
actualizados.   LOGIX5 SMART SOLUTIONS, S.L. B-86129699 se compromete a
respetar su confidencialidad en el tratamiento de sus datos personales, y
le informa que tiene derecho a ejercitar los derechos ARCO (derecho de
acceso, rectificación, cancelación y oposición) de sus datos personales
mediante solicitud escrita a ***@***.***, adjuntando fotocopia del
D.N.I.

"En la vida hay tres cosas que no dan marcha atrás: la palabra pronunciada,
la flecha lanzada y la oportunidad perdida, ¿vamos a dejarla escapar?"

rejas, 2022-03-13:
Thanks for your feedback. I might be of help with gettung the package.json in shape for linux (and mac) and the stuff that has to do with the electron builder etc.

I am not very familiar with the whole "arduino toolchain" and what has to be done there to get it ready for linux. Or is that a non-issue since linux support is working in the toolchain and only the isntaller needs to be fixed for linux?

So, having said that, here are more questions:
- you also use this repo for compiling the mac os dmg? because I cant find a script for that in the package.json
- what minimum version is required for node and npm to get this thing build?
- any reason why the electron version and other dependencies are stuck in very old and unspported version?
- the dist directory created when running the compiler task is placed outside the project dirs? on purpose or bug?

rejas, 2022-03-13:
Also: this seems to be a clone of blocklino (alas not a fork), just out of curiosity: what difference is to the original fork (broadly spoken) since this repo here is focused on otto diy?

cparrapa, 2022-03-13:
Good questions @rejas 

In regards to Linux we know very little how to work, I had one around but is soo slow that I could not use for debugging. 

1. This repo also can compile dmg you can see at the end of the package.json follow this https://docs.google.com/document/d/18_sVp2bEYxbGbeXeBdbZlCVgsqRWoC_0sH6FsysbwhA/edit?usp=sharing
2.3. This new versions for both npm and electron increase created errors in the compilations so we were afraid to update, so we are stuck there until is tested with new versions and we are sure that everything works.
4. The dist directory was like that from blocklino so not sure if is a good practice Thanks for the PR looks good！

logix5, 2022-03-13:
>    2. The dist directory was like that from blocklino so not sure if is a

Hi @rejas

The current  linux version is not using arduino-cli and the blockly is not
updated. https://github.com/OttoDIY/blockly/tree/versionlinux

The master branch is ready for creating the final version for windows and
Mac. We need to use the npm V12 version . With new version of npm the
release is not generated.

The idea is use the master branch to create the linux version too. In my
opinion , but i am not sure, only need to know how to call arduino-cli in
linux and place it in the folder and generate the linux application. This
file i think have the option to create the linux version
https://github.com/OttoDIY/blockly/blob/master/package.json.
Maybe using the current branch only need to create the linux version and
test it , because i think the arduino-cli is called  the same way as mac

the index.js have these lines:

var arduino_basepath = process.platform == 'win32' ? arduino_basepath =
'./compilation/arduino' : path.join(__dirname, '../../compilation/arduino')
var arduino_ide_cmd = process.platform == 'win32' ? 'arduino-cli.exe' :
arduino_ide_cmd = path.join(__dirname,
'../../compilation/arduino/arduino-cli')

As you can see these variable have different value is the Operating
system is win32 or other. If Linux call arduino-cli the same way to mac
only need to generate the linux version and test it

If you can test it would be perfect. I only have knowledge about blockly to
create blocks

Thanks in advance

BR

-- 
Óscar Ferruz Romero
*Socio fundador.*
*Tel. 620.402.168*

 C/ Artajona, 17.  28039 - Madrid.
www.logix5.com

www.twitter.com/Logix5_SS <http://www.twitter.com/logix5_SS>

www.facebook.com/logix5

www.instagram.com/logix5_smart_solutions

Conforme dispone el Reglamento (UE) 2016/679 del Parlamento Europeo y del
Consejo, de 27 de abril de 2016, relativo a la protección de las personas
físicas en lo referente al tratamiento de datos personales y a la libre
circulación de estos, le informamos que los datos personales facilitados y
recogidos en este e-mail y sus documentos adjuntos serán tratados e
incluidos automatizadamente en los ficheros de datos personales bajo
responsabilidad de LOGIX5 SMART SOLUTIONS, S.L. B-86129699, donde se
recogen y almacenan los datos personales con la finalidad exclusiva de
formar parte en los procesos de selección de personal, bolsa de trabajo,
contratación o gestiones relacionadas con temas laborales que se lleven a
cabo. El interesado otorga su consentimiento para el tratamiento de sus
datos personales con la finalidad anteriormente mencionada. En el supuesto
de producirse alguna modificación de sus datos personales, le solicitamos,
nos lo comunique por escrito con la única finalidad de mantenerlos
actualizados.   LOGIX5 SMART SOLUTIONS, S.L. B-86129699 se compromete a
respetar su confidencialidad en el tratamiento de sus datos personales, y
le informa que tiene derecho a ejercitar los derechos ARCO (derecho de
acceso, rectificación, cancelación y oposición) de sus datos personales
mediante solicitud escrita a ***@***.***, adjuntando fotocopia del
D.N.I.

"En la vida hay tres cosas que no dan marcha atrás: la palabra pronunciada,
la flecha lanzada y la oportunidad perdida, ¿vamos a dejarla escapar?"

rejas, 2022-03-14:
Took a swing at building a linux deb. Basically I only did:
- downloaded arduino cli for linux and put the file in "compilation/arduino"
- added a new npm script "build_linux": "builder --linux" for creating a deb (which worked with some small fix)
- installation of deb worked and blockly started
- connecting didnt work though:
`
Error: Command failed: /opt/OttoBlockly/compilation/arduino/arduino-cli compile --fqbn arduino:avr:nano sketch/sketch.ino Error: invalid path writing inventory file: inventory.yaml error: open inventory.yaml: permission denied time="2022-03-14T20:54:47+01:00" level=error msg="Error: invalid path writing inventory file: inventory.yaml error: open inventory.yaml: permission denied"`

rejas, 2022-03-14:
Also: Does the npm script "start" work for anyone? I only get a white window when running it due to [line 17](https://github.com/OttoDIY/blockly/blob/1211c27c70090fe60735d5eb512ee944dba07940/electronApp.js#L17)
where the path to the index.html file is created only for the packaged files and not for this usecase where the www-directory is in the same directory

logix5, 2022-03-14:
> changing that line to
> mainWindow.loadURL("file://" + path.join(__dirname, './www/index.html'))
> would make the "npm start" work but of course wouldnt work for the
> packaged installers.

Hi @rejas

I don´t know about npm, but your tests in linux are very good . You are
near of the final solution. Maybe the folder of the script doesn´t have
permission. This file sketch/sketch.ino must be find for the arduino-cli
and it will work right?

I hope you can progress, and let us know

Thanks so much!!

-- 
Óscar Ferruz Romero
*Socio fundador.*
*Tel. 620.402.168*

 C/ Artajona, 17.  28039 - Madrid.
www.logix5.com

www.twitter.com/Logix5_SS <http://www.twitter.com/logix5_SS>

www.facebook.com/logix5

www.instagram.com/logix5_smart_solutions

Conforme dispone el Reglamento (UE) 2016/679 del Parlamento Europeo y del
Consejo, de 27 de abril de 2016, relativo a la protección de las personas
físicas en lo referente al tratamiento de datos personales y a la libre
circulación de estos, le informamos que los datos personales facilitados y
recogidos en este e-mail y sus documentos adjuntos serán tratados e
incluidos automatizadamente en los ficheros de datos personales bajo
responsabilidad de LOGIX5 SMART SOLUTIONS, S.L. B-86129699, donde se
recogen y almacenan los datos personales con la finalidad exclusiva de
formar parte en los procesos de selección de personal, bolsa de trabajo,
contratación o gestiones relacionadas con temas laborales que se lleven a
cabo. El interesado otorga su consentimiento para el tratamiento de sus
datos personales con la finalidad anteriormente mencionada. En el supuesto
de producirse alguna modificación de sus datos personales, le solicitamos,
nos lo comunique por escrito con la única finalidad de mantenerlos
actualizados.   LOGIX5 SMART SOLUTIONS, S.L. B-86129699 se compromete a
respetar su confidencialidad en el tratamiento de sus datos personales, y
le informa que tiene derecho a ejercitar los derechos ARCO (derecho de
acceso, rectificación, cancelación y oposición) de sus datos personales
mediante solicitud escrita a ***@***.***, adjuntando fotocopia del
D.N.I.

"En la vida hay tres cosas que no dan marcha atrás: la palabra pronunciada,
la flecha lanzada y la oportunidad perdida, ¿vamos a dejarla escapar?"

casusbelli, 2022-03-15:
[..]
> We put them here https://github.com/OttoDIY/blocklyLinux then you need to remplace all the www folder from this repo to keep it updated.
>
> Ideally we want to merge to this repo so it is all together [...]

Thanks for the hints, I'll try replacing the code manually for starters.

Regarding OS X I was using last blockly/OS X releases, probably missing some build related dependencies from the developer setups. I might re-visit that and see if if can file some helpful tickets.

rejas, 2022-03-15:
Maybe we should open a discussion about "linux and mac installers" instead of discussing here everything?

cparrapa, 2022-03-15:
Good Idea @rejas 
One last thing you can have a look how @agomezgar managed to package Linux here https://github.com/agomezgar/OttoBlocklyLinux the only problem is that is not updated and lack of Arduino CLI

Install node.js
Clone the repository
Run npm install
If you want to test the software, run npm start
To compile, run npm run compiler

rejas, 2022-03-18:
> Regarding OS X I was using last blockly/OS X releases, probably missing some build related dependencies from the developer setups. I might re-visit that and see if if can file some helpful tickets.

I can confirm that the mac os version doesnt work for me neither (yeah, i have linux and mac laptops here as well as a windows machine so I am very eager to get blockly running on all three OS (windows 1.4.7 works fine, so next i slinux and mac :-)

Anyway, regarding mac, my observations are:

First the electron builder dependency has to be updated to at least v21 so that the dmg building process runs on macOS (Big Sur).

Secondly, the arduino cli is missing. Downloadable is only the single exectuable, but once the app runs and tries to compile/upload programs, it complains about a missing arduino_avt which has to be installed via
`./arduino-cli core install arduino:avt`

The files installed are in the gitignore and looking through the sourcecode here I find this commit:
https://github.com/OttoDIY/blockly/commit/412f1f23c429e86f1d4004350432523786b3539f

So, now I would like to know how those files can be re-generated at runtime. Any help @cparrapa  or @takujikawata-pr ?

logix5, 2022-03-19:
> 412f1f2

Thanks @rejas.  It would be perfect if otto blockly works in all operating
systems. when you are generating the release these steps are needed after
installing the arduino-cli:

I hope help you to generate all

   1.

   Install Arduino CLI

get arduino-cli.exe from https://github.com/arduino/arduino-cli/releases
and place it under compilation/arduino

   1.

   Execute following to install required tools

cd compilation\arduino

arduino-cli.exe core  install arduino:avr

arduino-cli core update-index

arduino-cli core  install ***@***.***

arduino-cli core install ***@***.***

arduino-cli core install ***@***.***

arduino-cli core install ***@***.***

arduino-cli core install ***@***.***

cd ..\..\

-- 
Óscar Ferruz Romero
*Socio fundador.*
*Tel. 620.402.168*

 C/ Artajona, 17.  28039 - Madrid.
www.logix5.com

www.twitter.com/Logix5_SS <http://www.twitter.com/logix5_SS>

www.facebook.com/logix5

www.instagram.com/logix5_smart_solutions

Conforme dispone el Reglamento (UE) 2016/679 del Parlamento Europeo y del
Consejo, de 27 de abril de 2016, relativo a la protección de las personas
físicas en lo referente al tratamiento de datos personales y a la libre
circulación de estos, le informamos que los datos personales facilitados y
recogidos en este e-mail y sus documentos adjuntos serán tratados e
incluidos automatizadamente en los ficheros de datos personales bajo
responsabilidad de LOGIX5 SMART SOLUTIONS, S.L. B-86129699, donde se
recogen y almacenan los datos personales con la finalidad exclusiva de
formar parte en los procesos de selección de personal, bolsa de trabajo,
contratación o gestiones relacionadas con temas laborales que se lleven a
cabo. El interesado otorga su consentimiento para el tratamiento de sus
datos personales con la finalidad anteriormente mencionada. En el supuesto
de producirse alguna modificación de sus datos personales, le solicitamos,
nos lo comunique por escrito con la única finalidad de mantenerlos
actualizados.   LOGIX5 SMART SOLUTIONS, S.L. B-86129699 se compromete a
respetar su confidencialidad en el tratamiento de sus datos personales, y
le informa que tiene derecho a ejercitar los derechos ARCO (derecho de
acceso, rectificación, cancelación y oposición) de sus datos personales
mediante solicitud escrita a ***@***.***, adjuntando fotocopia del
D.N.I.

"En la vida hay tres cosas que no dan marcha atrás: la palabra pronunciada,
la flecha lanzada y la oportunidad perdida, ¿vamos a dejarla escapar?"

rejas, 2022-03-19:
Thanks for the info @logix5 

I did those steps already (for mac). But I was wondering if those things can happen during building the installer?

takujikawata-pr, 2022-03-19:
Hi,
Something from Mac build! I hope this may help some! I just added 1.4.7 installer in the release (https://github.com/OttoDIY/blockly/releases/download/v1.4.7/ottoblockly-1.4.7.dmg)

Mac build is still kind of experimental, so I think I need further adjustment to the build steps.
F.Y.I, following is my memo for Mac build:
https://docs.google.com/document/d/18_sVp2bEYxbGbeXeBdbZlCVgsqRWoC_0sH6FsysbwhA/edit?usp=sharing
(If Mac release doesn't work, I'd like to know how it fails. I'll then analysis the problem and will fix!)
 
I guess Linux build can also be done something similar. I'll try Linux build, too.

logix5, 2022-03-19:
Hi all!

The packages of every board must be installed using arduino cli before
packing to create the release.  These folder are created and then you must
create using the npm the final release with these folders

I hope the linux and mac version can be fixed and works perfectly

thanks

-- 
Óscar Ferruz Romero
*Socio fundador.*
*Tel. 620.402.168*

 C/ Artajona, 17.  28039 - Madrid.
www.logix5.com

www.twitter.com/Logix5_SS <http://www.twitter.com/logix5_SS>

www.facebook.com/logix5

www.instagram.com/logix5_smart_solutions

Conforme dispone el Reglamento (UE) 2016/679 del Parlamento Europeo y del
Consejo, de 27 de abril de 2016, relativo a la protección de las personas
físicas en lo referente al tratamiento de datos personales y a la libre
circulación de estos, le informamos que los datos personales facilitados y
recogidos en este e-mail y sus documentos adjuntos serán tratados e
incluidos automatizadamente en los ficheros de datos personales bajo
responsabilidad de LOGIX5 SMART SOLUTIONS, S.L. B-86129699, donde se
recogen y almacenan los datos personales con la finalidad exclusiva de
formar parte en los procesos de selección de personal, bolsa de trabajo,
contratación o gestiones relacionadas con temas laborales que se lleven a
cabo. El interesado otorga su consentimiento para el tratamiento de sus
datos personales con la finalidad anteriormente mencionada. En el supuesto
de producirse alguna modificación de sus datos personales, le solicitamos,
nos lo comunique por escrito con la única finalidad de mantenerlos
actualizados.   LOGIX5 SMART SOLUTIONS, S.L. B-86129699 se compromete a
respetar su confidencialidad en el tratamiento de sus datos personales, y
le informa que tiene derecho a ejercitar los derechos ARCO (derecho de
acceso, rectificación, cancelación y oposición) de sus datos personales
mediante solicitud escrita a ***@***.***, adjuntando fotocopia del
D.N.I.

"En la vida hay tres cosas que no dan marcha atrás: la palabra pronunciada,
la flecha lanzada y la oportunidad perdida, ¿vamos a dejarla escapar?"

rejas, 2022-03-19:
> 

Yeah, your memo is quite accurate to how I managed to make a dmg on my mac. 

Am I correct, that there is no automatic release-process that puts the exe and dmg installers into the github? You do it manually?

cparrapa, 2022-04-20:
Hi @rejas 
Sorry for the late reply here, I have been busy and then the holidays...

I am not sure for Mac but for windows I just do a **npm run publier** this automatically creates the new release with the new version number and puts the exe installers into the github, but this requires the token for the repo which you should be able to create as member of the repo.

All PR merged, what is left to put in sync all OS, specially the linux one that we are missing?

closed 2024-06-30 by cparrapa