OttoDIY/blockly #171: Otto wheels didn't stop

lsprandel, 2022-06-25:
The function to move forward or backward didn't stop after the delay.
For a symple program like that: The Otto wheels need to move forward by 1 second than stop by 1 second.
 
void loop() {
    motorControl (-45, -45, 1 );
    delay(1*1000);
}

In this exemple, Otto will run forever. Ther is no intuitive way to stop the Otto.

Maybe to change this situation, We can change the motorControl function.

Actual:
void motorControl(int rightSpeed, int leftSpeed, int stepDelay) {
rightServo.write(90 + rightSpeed);  leftServo.write(90 - leftSpeed);
delay(stepDelay*1000);} 

Proposition:
void motorControl(int rightSpeed, int leftSpeed, int stepDelay) {
rightServo.write(90 + rightSpeed);  leftServo.write(90 - leftSpeed);
delay(stepDelay*1000);rightServo.write(90);leftServo.write(90);}

[image]

cparrapa, 2022-06-25:
Good point @lsprandel  we will update that in the next release

Thanks! let us know if you find more improvements

cparrapa, 2022-09-27:
Fixed in https://github.com/OttoDIY/blockly/commit/03860be0e943fb8cb3b3399f83ff19866b9a86ab

closed 2022-09-30 by cparrapa