OttoDIY/blockly #173: library <PlayRtttl.hpp>

ebaltodano10, 2022-09-29:


ebaltodano10, 2022-09-29:
is:issue is:open  the library <PlayRtttl.hpp> was deleted, and now the buzzer with song not sound

logix5, 2022-09-29:
Hi  in the repository i can see the library

https://github.com/OttoDIY/blockly/tree/master/compilation/arduino/userlibs/libraries/PlayRtttl-master/src

maybe in the new version the library was not deliver. I have my local
version and it is works

-- 
Óscar Ferruz Romero
*Socio fundador.*
*Tel. 620.402.168*

 C/ Artajona, 17.  28039 - Madrid.
www.logix5.com

www.twitter.com/Logix5_SS <http://www.twitter.com/logix5_SS>

www.facebook.com/logix5

www.instagram.com/logix5_smart_solutions

Conforme dispone el Reglamento (UE) 2016/679 del Parlamento Europeo y del
Consejo, de 27 de abril de 2016, relativo a la protección de las personas
físicas en lo referente al tratamiento de datos personales y a la libre
circulación de estos, le informamos que los datos personales facilitados y
recogidos en este e-mail y sus documentos adjuntos serán tratados e
incluidos automatizadamente en los ficheros de datos personales bajo
responsabilidad de LOGIX5 SMART SOLUTIONS, S.L. B-86129699, donde se
recogen y almacenan los datos personales con la finalidad exclusiva de
formar parte en los procesos de selección de personal, bolsa de trabajo,
contratación o gestiones relacionadas con temas laborales que se lleven a
cabo. El interesado otorga su consentimiento para el tratamiento de sus
datos personales con la finalidad anteriormente mencionada. En el supuesto
de producirse alguna modificación de sus datos personales, le solicitamos,
nos lo comunique por escrito con la única finalidad de mantenerlos
actualizados.   LOGIX5 SMART SOLUTIONS, S.L. B-86129699 se compromete a
respetar su confidencialidad en el tratamiento de sus datos personales, y
le informa que tiene derecho a ejercitar los derechos ARCO (derecho de
acceso, rectificación, cancelación y oposición) de sus datos personales
mediante solicitud escrita a ***@***.***, adjuntando fotocopia del
D.N.I.

"En la vida hay tres cosas que no dan marcha atrás: la palabra pronunciada,
la flecha lanzada y la oportunidad perdida, ¿vamos a dejarla escapar?"

cparrapa, 2022-11-04:
thanks @ebaltodano10  now there is a new release with the library back

closed 2022-11-04 by cparrapa