OttoDIY/blockly #197: ¿Soporte para mas tarjetas?

NeftaliVenancio, 2024-01-12:
se prevee agregar soporte para más tarjetas, por ejemplo la raspberry pi pico?

sfranzyshen, 2024-05-05:
>> se prevee agregar soporte para más tarjetas, por ejemplo la raspberry pi pico?
>> Is it planned to add support for more cards, for example the raspberry pi pico?

blockly está diseñado para usar la biblioteca Arduino OttoDIYLib... sí, hay planes futuros en desarrollo para admitir el pico... no hay una estimación de tiempo.

blockly is built to use the Arduino OttoDIYLib library ... yes there are future plans under development to support the pico ... there is no estimate of time.

https://github.com/OttoDIY/OttoDIYLib/tree/devel

sfranzyshen, 2024-05-05:
Another issue is the 3d model will need to be remixed to support a Pico shield ...
Otro problema es que será necesario remezclar el modelo 3D para admitir un escudo Pico...